TheKikGen/stm32-tkg-hid-bootloader #1: Corrupt flash upload

TyberiusPrime, 2020-07-08:
First off, thanks for writing this bootloader, it is very clearly structured and hackable and who doesn't like it's tiny size, both in binary and in code!

Unfortunatly, on my blue pills, it fails to flash correctly and my binaries don't run.

I've created 'combined' binaries (bootloader-padded-to-4096 bytes)/blinky,
and in those the board does blink correctly.

Now I enter the bootloader, flash the same blinky binary I used to create the combined binary,
and it does no longer blink.

I've retrieved the flash and I found, that the bootloader-flashed-blinky
- the first 4096 bytes are identical (as they should)
- the bootloader-flashed binary is offset by one byte 'to the right'. There is a 0x90 byte *before* the binary 
- correcting for this, there starts to be occasional differences at byte 4160
- after byte 4440 basically every byte is different.

(all of this is with bootloader_only/binaries/tkg_hid_generic_pc13.bin as well as with a version I stripped down to 'enter on double reset, don't touch peripherals at all' - I initially thought the issue was one of the MCU being in a state where my rust binary wouldn't initialize it correctly).

Now the corruption could be a timing issue - I'm flashing from linux. I don't see a CRC in the flash code, correct?

But the of by one - if I shift my combined binary by that 0x90, it stops working as well....

Sorry for the wall of text.
Hopefully you have some insights for me :).

TheKikGen, 2020-07-08:
Hi Florian,

Yes. probably a timing issue.  I tested on Linux but not so deeply, because
for various reasons, my main development platform is under WIN10.
Can you test on WIN ?

No CRC but I'm seriously thinking about it !
I probably should reduce the size of a block and add  e.g. a XOR CRC, and
add a handshake.

Franck

TheKikGen, 2020-07-10:
I made an update of TKG-FLASH including an increasing delay between packet from 500 uS to 1 mS.

closed 2021-01-18 by TheKikGen