TheKikGen/stm32-tkg-hid-bootloader #3: Wait for firmware upload too short for Arduino IDE

riban-bw, 2020-08-12:
I have installed the stm32-tkg-hid-bootloader on a STM32F103C8 using the tkg_hid_generic_pc13.bin firmware. This works fine when double pressing the reset or asserting BOOT1 but fails when the module is reset by toggle of USB Serial DTR as used by Arduino upload routine. This is an issue when using Roger Clark's Arduino_STM32 libraries and associated "HID bootloader 2.0"  upload method. (Note: I need to use libraries provided by Roger Clark that are not available elsewhere.)

I have found that changing the dynamics of the bootloader improves the behaviour without obvious significant side-effects:

- Increase the "2nd Waiting round for an eventual bootloader START cmd" from 1s to 2s
- Shorten the double press period from 3s to 2s (so that overall start time is not extended)

I have found the bootloader works reliably with these changes and there is still (more than enough) time to manually trigger bootloader with double press of reset button.

Note: To use DTR reset the user code must implement USB Serial port, e.g. USBCompositeSerial.

riban-bw, 2020-08-12:
Actually - I have adjusted various timings and wonder whether they may all be a bit long (apart from the wait for START cmd). I understand the bootloader needs to be robust and cope with slow USB enumeration, etc. but it takes about 7s to reach user code which I believe could be reduced to approx. 4s (although you may have other, real-life data which contradicts this).

TheKikGen, 2020-09-28:
Interesting.    But as you mentioned, that works for your context, but maybe not for many others !
I played a lot with timing...I will give a try to yours.  Now that you can increase the waiting time on the TKG-Flash command line (so in the Arduino IDE flash command).

closed 2021-01-26 by TheKikGen