TheKikGen/stm32-tkg-hid-bootloader #4: Error flashing

condor13, 2020-08-12:
Hello.
I`m having a little issue using the flasher. I flashed the bootloader using stlink successfully. Windows detects HID device, flasher detects hid device but gives "Error while sending firmware data to HID device." 
I have the bluepill pc13 board 64k flash. Do i have to make mods to the board like [here](https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Flashing-Bootloader-for-BluePill-Boards).
Thanks. 
Teodor.

TheKikGen, 2020-08-12:
Hello,

What operating system are you using ?

condor13, 2020-08-12:
Windows 7 x64 pro.

condor13, 2020-08-13:
ok. your reply implied that i should try another os. so i used win10 pro. worked on first try. My norns loves this already :+1:

closed 2020-08-13 by condor13