TheKikGen/stm32-tkg-hid-bootloader #5: tkg-flashing unsuccesful on linux, but st-flash works

jalkjaer, 2020-09-28:
I have flashed the boot loader succesfully using st-flash
Device shows up as, Bus 001 Device 034: ID 1209:beba Generic serasidis.gr STM32 HID Bootloader
Using tkg-flash (here a dump of 2.5.0 img)
```
sudo ./tkg-flash umk4x4.stm32f103c8-64k-o2.bin
+-----------------------------------------------------------------------+
|            TKG-Flash v2.3 STM32F103 HID Bootloader Flash Tool         |
|                     High density device support.                      |
|       (c) 2020 - The KikGen Labs     https://github.com/TheKikGen     |
+-----------------------------------------------------------------------+

> Firmware file size is 53924 bytes.
> Searching for [1209:BEBA] HID device...|
> [1209:BEBA] HID device found !
> Flashing firmware :

  [++++++++++++++++++++++++++++++++++++++++++++++++++]   54272 bytes sent.

> Firmware flashing done !
  53 sectors written.
> End of firmware update.
```
But the device is stuck in the bootloader no matter how many reset clicks/power cycles I do (havn't fiddled with jumpers).

I was however able to flash the umk4x4.stm32f103cb-128k-o3.bin using st-flash to the location right after the boorloader
```st-flash --flash=128k  write umk4x4.stm32f103cb-128k-o3.bin 0x08001000```
and that works. 

Unsure where the problem lies (tk-flash/boot-loader)

dmesg does show some usb probe problem when in HID mode  - but I suspect that is unrelated and benign
```
usb 1-1: new full-speed USB device number 77 using xhci_hcd
usb 1-1: string descriptor 0 read error: -22
usb 1-1: New USB device found, idVendor=1209, idProduct=beba, bcdDevice= 3.00
usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
hid-generic 0003:1209:BEBA.0031: hiddev0,hidraw0: USB HID v1.11 Device [HID 1209:beba] on usb-0000:00:14.0-1/input0
```

jalkjaer, 2020-09-28:
Overwrote the old bootloader
The output is now:
```
Sep 28 23:07:21 jalap kernel: usb 1-1: new full-speed USB device number 57 using xhci_hcd
Sep 28 23:07:21 jalap kernel: usb 1-1: New USB device found, idVendor=1209, idProduct=beba, bcdDevice= 3.00
Sep 28 23:07:21 jalap kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Sep 28 23:07:21 jalap kernel: usb 1-1: Product: STM32F HID Bootloader
Sep 28 23:07:21 jalap kernel: usb 1-1: Manufacturer: The KiKGen Labs
Sep 28 23:07:21 jalap kernel: hid-generic 0003:1209:BEBA.004F: hiddev0,hidraw0: USB HID v1.11 Device [The KiKGen Labs STM32F HID Bootloader] on usb-0000:00:14.0-1/input0
Sep 28 23:07:23 jalap kernel: usb 1-1: USB disconnect, device number 57
Sep 28 23:07:26 jalap kernel: usb 1-1: new full-speed USB device number 58 using xhci_hcd
Sep 28 23:07:26 jalap kernel: usb 1-1: New USB device found, idVendor=2912, idProduct=1970, bcdDevice= 1.00
Sep 28 23:07:26 jalap kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 28 23:07:26 jalap kernel: usb 1-1: Product: USB MIDIKliK 4x4
Sep 28 23:07:26 jalap kernel: usb 1-1: Manufacturer: TheKiKGen Labs
Sep 28 23:07:26 jalap kernel: usb 1-1: SerialNumber: 07DA0908
```
Double click on reset to get into bootloader
```
Sep 28 23:07:54 jalap kernel: usb 1-1: USB disconnect, device number 58
Sep 28 23:07:54 jalap kernel: usb 1-1: new full-speed USB device number 59 using xhci_hcd
Sep 28 23:07:54 jalap kernel: usb 1-1: New USB device found, idVendor=1209, idProduct=beba, bcdDevice= 3.00
Sep 28 23:07:54 jalap kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Sep 28 23:07:54 jalap kernel: usb 1-1: Product: STM32F HID Bootloader
Sep 28 23:07:54 jalap kernel: usb 1-1: Manufacturer: The KiKGen Labs
Sep 28 23:07:54 jalap kernel: hid-generic 0003:1209:BEBA.0050: hiddev0,hidraw0: USB HID v1.11 Device [The KiKGen Labs STM32F HID Bootloader] on usb-0000:00:14.0-1/input0
```
Using `tkg-flash <image>` is however still unsuccessful. The flashing reports success like before, but the device is stuct in bootloader mode after reset / power-cycle
So i have to "st-link" flash the firmware to addr 0x8001000

One of my boards does exibit some odd behavior when plugging it in:
```
Sep 28 23:23:54 jalap kernel: usb 1-1: new full-speed USB device number 83 using xhci_hcd
Sep 28 23:23:53 jalap kernel: usb 1-1: device descriptor read/64, error -71
Sep 28 23:23:53 jalap kernel: usb 1-1: device descriptor read/64, error -71
Sep 28 23:23:54 jalap kernel: usb 1-1: new full-speed USB device number 84 using xhci_hcd
Sep 28 23:23:54 jalap kernel: usb 1-1: device descriptor read/64, error -71
Sep 28 23:23:54 jalap kernel: usb 1-1: device descriptor read/64, error -71
Sep 28 23:23:54 jalap kernel: usb usb1-port1: attempt power cycle
Sep 28 23:23:55 jalap kernel: usb 1-1: new full-speed USB device number 85 using xhci_hcd
Sep 28 23:23:55 jalap kernel: usb 1-1: device descriptor read/8, error -71
Sep 28 23:23:55 jalap kernel: usb 1-1: device descriptor read/8, error -71
Sep 28 23:23:55 jalap kernel: usb 1-1: new full-speed USB device number 86 using xhci_hcd
Sep 28 23:23:55 jalap kernel: usb 1-1: device descriptor read/8, error -71
Sep 28 23:23:55 jalap kernel: usb 1-1: device descriptor read/8, error -71
Sep 28 23:23:55 jalap kernel: usb usb1-port1: unable to enumerate USB device
Sep 28 23:23:58 jalap kernel: usb 1-1: new full-speed USB device number 87 using xhci_hcd
Sep 28 23:23:58 jalap kernel: usb 1-1: New USB device found, idVendor=2912, idProduct=1970, bcdDevice= 1.00
Sep 28 23:23:58 jalap kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 28 23:23:58 jalap kernel: usb 1-1: Product: BluePill 1
Sep 28 23:23:58 jalap kernel: usb 1-1: Manufacturer: TheKiKGen Labs
Sep 28 23:23:58 jalap kernel: usb 1-1: SerialNumber: 07DA0908
```
After this initial "boot" issue, switching between config / bootloader and resetting works perfectly, so I assume this is hardware specific, potentially some initial timing variation between boards.

TheKikGen, 2021-01-26:
Last version should correct many syncs and usb issues....

closed 2021-01-26 by TheKikGen