TheKikGen/stm32-tkg-hid-bootloader #6: TKG-Flash not working on Windows 7

GitMauri, 2021-01-29:
Hi KikGenius

Thanks again for your work. Great HID Bootloader. As you mentioned it needs the IDE integration to be tuned in order to get rid of the ide error

`error while uploading: missing 'upload.parms.quiet' configuration parameter`

And there is a **problem** trying to Flash by command line a compiled bin sketch using tkg-flash (3.1) on **Windows 7 64** machines

```
+-----------------------------------------------------------------------+
|             TKG-Flash v3.1 STM32F103 HID Bootloader Flash Tool        |
|                     High density device support.                      |
|       (c) 2020 - The KikGen Labs     https://github.com/TheKikGen     |
+-----------------------------------------------------------------------+
Build : 3.210120.1703

> Searching for [1209:BEBA] HID device...|
> [1209:BEBA] HID device found !
> Firmware file size is 16212 bytes.
> Flashing firmware :
  INFO - Informations reported by the bootloader :
      Firmware version       : 0310
      MCU Flash memory size  : 256 K (high density device)
      Page size              : 2048 bytes
      Page offset            : 2 pages
      Flash base address     : 0x08001000
  START command sent successfully.

  [..................................................]
  ** Error while sending firmware data to the HID device.
> TKG-Flash end.
```

**The same procedure succeed on Windows 10**

```

+-----------------------------------------------------------------------+
|             TKG-Flash v3.1 STM32F103 HID Bootloader Flash Tool        |
|                     High density device support.                      |
|       (c) 2020 - The KikGen Labs     https://github.com/TheKikGen     |
+-----------------------------------------------------------------------+
Build : 3.210129.0246

> Searching for [1209:BEBA] HID device...|
> Searching for [1209:BEBA] HID device...-
> [1209:BEBA] HID device found !
> Firmware file size is 16196 bytes.
> Flashing firmware :
  INFO - Informations reported by the bootloader :
      Firmware version       : 0310
      MCU Flash memory size  : 256 K (high density device)
      Page size              : 2048 bytes
      Page offset            : 2 pages
      Flash base address     : 0x08001000
  START command sent successfully.

  [++++++++++++++++++++++++++++++++++++++++++++++++++]   16384 bytes sent.
  No checksum error.

> Firmware flashing done !
  16 sectors written.
> TKG-Flash end.
```

Thanks again

Maurizio

TheKikGen, 2021-01-29:
Hi Maurizio.

Happy to see that Win10 is working well now.
I think again, it is a timing problem, but it could be also linked with usb caching on the host on your Win7 platform.

Install USBDEVIEW http://www.nirsoft.net/utils/usb_devices_view.html.
Unplug your board from usb and in the option menu , check "Display disconnected devices" or use CTRL+F8.
Click on the VendorId column header to sort vendor ids.
Remove all pairs (1209, beba) by right clicking and choose "uninstall selected device" in the popup menu.
That will clean your usb cache from all previous installation.
Plug again your board and try to upload something...

TheKikGen, 2021-01-29:
If the previous usb cleaning method still doesn't work on Win7, try this new tkg-flash version attached.Build : 3.210129.1313

[tkg-flash.zip](https://github.com/TheKikGen/stm32-tkg-hid-bootloader/files/5893762/tkg-flash.zip)

TheKikGen, 2021-01-29:
This release should solve the issue : https://github.com/TheKikGen/stm32-tkg-hid-bootloader/releases/tag/V3.11

GitMauri, 2021-01-30:
Hi KikGenius

Sorry for my late reply. Yesterday I went out.

Your new release now can be used on Windows 7 in the CMD prompt to flash the compiled sketch.bin !!!!! It works,  GREAT!!!!!!

But there is still the problem with the IDE

`missing 'upload.parms.quiet' configuration parameter`

I know it is a lot of work to tune up everything keeping compatibility with OS s and releases.... 

Thanks again

TheKikGen, 2021-01-30:
check if the parameter in Arduino/hardware/Arduino_STM32/STM32F1/platform.txt is correctely labelled 

tools.tkg_hid_upload.upload.params.quiet=n

GitMauri, 2021-01-31:
Hi KikGenius

Yes those parameters were ok. 

But I double checked the board references... When I copied the board.txt modifications from the old read.me file I didn't check for miss spelling... I just modified the name of the board and the name of the .ld file...
The error was already visible in my Serasisid repo post ...     ...UploadMethod2K....

```
genericSTM32F103R.menu.upload_method.HIDUploadMethod2K=TKG-HID bootloader
genericSTM32F103R.menu.upload_method.HIDUploadMethod2K.upload.tool=tkg-hid
genericSTM32F103R.menu.upload_method.HIDUploadMethod2K.build.upload_flags=-DSERIAL_USB -DGENERIC_BOOTLOADER
genericSTM32F103R.menu.upload_method.HIDUploadMethod2K.build.vect=VECT_TAB_ADDR=0x8002000
genericSTM32F103R.menu.upload_method.HIDUploadMethod2K.build.ldscript=ld/hid_bootloader_tkg_r.ld
```

just got rid of the extra text (2k) and it works... 

However your new read.me file for 3.11 is already ok, so it is safe to copy and paste, and to change the board name if needed. Every thing is very well explained and in a way very didactic !!!!

Now I'm testing it together with Arprus USB composite and so far it is working ok... testing mass storage... 

Many many many thanks.

closed 2021-01-31 by GitMauri

TheKikGen, 2021-01-31:
:-)