TheKikGen/stm32-tkg-hid-bootloader #8: CH32 devices?

MickGyver, 2021-05-22:
Can anything be done to make CH32F103C8T6 devices work? I have tried several bootloaders but have only managed to get Roger Clarke's bootloader to work, the other ones, including this one gives the error "request for usb hid device descriptor failed". HID device firmware also works fine, device is detected properly in Windows 10.

TheKikGen, 2021-05-27:
Hi.  This is probably not linked with software but with hardware.
Did you check USB resistors ?
Did you check on several computer ?
Are you using a "Bluepill" ?

MickGyver, 2021-05-29:
Thanks for the reply! The USB resistor is 1.5K. Some HID joystick code I tested works fine on the board when uploaded using ST-Link, the bootloader by Roger Clark (stm32duino) also works fine, but I don't want to use that since it requires drivers.

TheKikGen, 2021-06-02:
If you are able to, you could increase waiting time in the firmware here : https://github.com/TheKikGen/stm32-tkg-hid-bootloader/blob/f403c39b88b0ca438e243d821b475cf1c31bf38f/F1/Src/main.c#L101

MickGyver, 2021-06-02:
Thanks for the help, I just tried it but alas, the problem remains.